STMicroelectronics/stm32-mw-usb-host #1: Event Queue Full

GiulioDallaVecchia, 2021-04-13:
**Describe the set-up**
 * STM32H735G-DK
 * STM32CubeIDE

**Describe the bug**
I'm using the USB Host library with FreeRTOS implementation with CMSIS OS2. Each time I've tried to insert anUSB Stick, the USB Core message queue reach the errQueue_FULL error. 

**How To Reproduce**
1. Create a project with FreeRTOS

2. Added USB Host Library last version (v3.3.5)

3. Insert a USB stick and keep under monitoring the osMessageQueuePut function call

**Screenshots**
[image]

ALABSTM, 2021-04-21:
Hi @GiulioDallaVecchia,

We will try to analyze your report as soon as possible and get back to you as soon as we have an answer. Thank you for your patience.

With regards,

RKOUSTM, 2021-07-06:
Hi @GiulioDallaVecchia,

Thank you for this report. 

If the issue is systematically reproducible on your side, could you please share the whole project so that we can reproduce the same behavior and give you an answer.

With regards,

ALABSTM, 2021-08-02:
Hi @GiulioDallaVecchia,

We hope you are fine. Any project to share or at least an IOC file? Thank you.

With regards,

RKOUSTM, 2021-08-13:
Hi @GiulioDallaVecchia,

Any new comments to move this discussion forward? 

With regards,

RKOUSTM, 2021-08-16:
Hi @GiulioDallaVecchia,

Please allow me to close this thread as no feedback has been received. You can reopen it if you have anything else relevant to add regarding this topic. 

With regards,

GiulioDallaVecchia, 2021-09-01:
Hi @RKOUSTM,

I have a ready project. What is the best solution (or platform) for you in order to share the project?

ALABSTM, 2021-12-22:
Hi @GiulioDallaVecchia,

I hope you are doing well. I want to inform you that I run the project you sent but could not yet reproduce the issue. I will keep you updated.

With regards,

ALABSTM, 2022-04-25:
Hi @GiulioDallaVecchia,

The issue you described has been logged internally and escalated to our USB team. We will get back to you once we have a feedback.

With regards,

ALABSTM, 2022-04-25:
ST Internal Reference: 126912

ALABSTM, 2023-11-22:
Hi @GiulioDallaVecchia,

I hope you are fine. Back to you about this point you submitted a time ago.

Our development teams could reproduce the behavior you described. However, according to them, the queue size is rather a **design choice**, than a limitation. They suggest that the default size of the queue could be **overridden at user's application level**.

Please allow me to close this point in this case. Thank you for your comprehension and for your patience.

Below some feedback from the development teams' observations:
* The queue is used to block/unblock the `USBH_Process_OS()` thread.
* Several functions executing within the context of the USB ISR put events on the queue.
* The `USBH_Process_OS()` thread itself puts events on the queue to keep itself running and process state transitions.
* The events are not used in anyway and seem to serve no other purpose than unblocking `USBH_Process_OS()`.
* The brute force approach of increasing the queue size to a large value eliminates the `errQueue_FULL` issue yet the app behavior is the same. The device is successfully enumerated in both cases.

With regards,

closed 2023-11-22 by ALABSTM