STMicroelectronics/stm32-mw-usb-host #5: Overrun when interface ep_ix >= USBH_MAX_NUM_ENDPOINTS

thornley-permaconn, 2022-01-05:
Referencing: -

https://github.com/STMicroelectronics/stm32_mw_usb_host/blob/70f090d8ff1714c6eb9a48c3d59f2065cb0a6636/Core/Src/usbh_ctlreq.c#L479

Ep_Desc[ep_ix] will overrun memory if device plugged in with endpoints greater than the preallocated USBH_MAX_NUM_ENDPOINTS array (within the interface). Protection is present for if_ix (USBH_MAX_NUM_INTERFACES) but not ep_ix.

thornley-permaconn, 2022-01-05:
Thanks @ALABSTM

ALABSTM, 2022-02-28:
Hi @thornley-permaconn,

Thank you for having reported this point. Actually, it looks like @Defonceuse submitted a pull-request with a fix proposal. The point will be forwarded to our development teams. I will get back to you as soon as I have their feedback.

May I ask you whether you noticed the point just by reviewing the code or whether you actually experienced a failure due to this implementation? Thank you in advance for your reply.

With regards,

thornley-permaconn, 2022-02-28:
Hi @ALABSTM, 

I was just running a proof of concept. It is not uncommon to have a large number of endpoints on cellular devices for example. 

No rush on my part, I just wanted to do the right thing by reporting it.

ALABSTM, 2022-03-01:
Hi @thornley-permaconn,

Thank you very much for the time taken to report. Your point has been submitted to our development teams, as already said. If confirmed, it will be logged into our internal database to have it fixed. I will keep you updated.

With regards,

ALABSTM, 2022-03-04:
ST Internal Reference: 123858

sburgh01, 2022-10-20:
As further Info. This made problemes when switching devices with different number of Endpoints, because the content is never cleared and then the Endpoints are misinterpreted on the next Device.

closed 2023-03-29 by ALABSTM