STMicroelectronics/stm32-mw-usb-host #8: v3.4.1 fail MSC with usb stick. v3.4.0 is ok

GLASS-FR, 2022-07-23:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * custom board stm32f765 and stm32g0
 * Cube IDE 1.10.1
 * cube mx 6.6.1

**Describe the bug**
Before, using usb host lib v.3.4.0 we can use usb stick with no problem.
On stm32f765, upgrading hal to 1.17.0 we got also usb host lib v.3.4.1.
After lot of croised tests it appear that reverting only this lib to v3.4.0 give us a functionnal MSC USB HOST.

See :
https://community.st.com/s/question/0D53W00001fPT6NSAW/stm32g0b1-usb-host-msc-flash-drive-not-working

**How To Reproduce**
 Cube mx generated code for stm32f765 (seems exactly the same for other user with stm32g0) with latest hal (1.17.0 for F7).

Can be reproduced just plugging a usb stick (4gb fat32 ).
And the v.3.4.1 lib go to infinite oop.

GLASS-FR, 2022-07-24:
Other users hit same issue .
Give a look here 👍 
[y.st.com/s/question/0D53W00001eaR69SAE/fatfsusbdisk-example-with-usbfs-does-not-work-when-an-usb-stick-is-connected-the-usbhmscprocess-never-gets-further-then-mscreadinquiry-any-hints](https://community.st.com/s/question/0D53W00001eaR69SAE/fatfsusbdisk-example-with-usbfs-does-not-work-when-an-usb-stick-is-connected-the-usbhmscprocess-never-gets-further-then-mscreadinquiry-any-hints)

ABOUSTM, 2022-08-12:
Hello GLASS-FR,
Thanks for reporting the issue, the fix is part of Host lib v3.5.0 available under ST github

Regards,
Aymen

GLASS-FR, 2022-10-01:
@ABOUSTM  and @ALABSTM : please its urgent to update : 
1)Release note  : v3.4.1 MSC regression solved by v3.5.0
2) All Hal and cube package using 3.4.1=>3.5.0 with release note clearly explaining msc regression solved by v3.5.0 usb host lib
Hope you will can do something rapidement because we are numerous to fall in this trap...

hgscania, 2023-01-03:
Hi. I have a problem in usb host with stm32f429ii. In usb_core.c line1145 ,'USBH_Get_StringDesc' return busy and in usbh_msc.c line 459 (MSC_REQUEST_SENSE) scsi_status=USBH_FAIL and device not ready. What should I do?

ALABSTM, 2023-05-15:
Hi @GLASS-FR,

Thank you once more for having reported this point.

Regarding the upgrade of USB Host library to the latest version in all firmware packages, there is currently no plan for such update concerning certain series. We will keep you updated should there be any change.

In the meanwhile, you can easily perform the integration into your package by yourself as newer versions are backward compatible. Thank you for your comprehension.

With regards,

closed 2023-05-15 by ALABSTM