STMicroelectronics/stm32-mw-usb-host #12: Why MSC_REQUEST_SENSE results in No error and  HOST_USER_CLASS_ACTIVE

NimaAgm, 2022-10-23:
I am developing Stm32f730 USB host application.
For testing my application I have connected a turned off android phone to my USB port.
device gets enumerated as a MSC Device
This is the output of Enumeration:

USB Device Connected
USB Device Reset Completed
PID: 2eb7h
VID: 22b8h
Address (#1) assigned.
Manufacturer : motorola
Product : moto g(9) play
Serial Number : ZY3282Q2V5
Enumeration done.
This device has only 1 configuration.
Default configuration set.
Switching to Interface (#0)
Class    : 8h
SubClass : 6h
Protocol : 50h
MSC class started.
Number of supported LUN: 1
LUN #0: 
Inquiry Vendor  : Linux   
Inquiry Product : File-Stor Gadget
Inquiry Version : 0419
Sense Key  : 2
Additional Sense Code : 3a
Additional Sense Code Qualifier: 0
Sense Key  : 2
Additional Sense Code : 3a
Additional Sense Code Qualifier: 0
ERROR: USB Disk is not ready!

**Why when device is not ready This below code executes ?**
phost->pUser(phost, HOST_USER_CLASS_ACTIVE);
device surely has a problem and it is not ready 
This code means:
<html>
<body>

2 | 3A | 00 | Not Ready - medium not present
-- | -- | -- | --

</body>
</html>

https://en.wikipedia.org/wiki/Key_Code_Qualifier

ALABSTM, 2023-04-04:
Hi @NimaAgm,

Please excuse this delayed reply. Your request has been forwarded ot our development teams. We will get back to you as soon as we get updates.

With regards,

ALABSTM, 2023-04-18:
Hi @NimaAgm,

You mentioned the STM32F7 series. Are you using the USB Host library in the STM32CubeF7 firmware or the one published in this repository?

Please note the library version in this repository **3.5.1**, while the one in the STM32CubeF7 firmware is **3.4.1**.

With regards,

NimaAgm, 2023-04-19:
I have used library version 3.5.0

ALABSTM, 2023-04-19:
ST Internal Reference: 150224

NimaAgm, 2023-07-18:
Could you provide feedback on this issue?

ABOUSTM, 2023-10-02:
Hello NimaAgm,
Change request acknowledged and to be part of next release in coming weeks

Regards

ALABSTM, 2023-12-20:
Hi @NimaAgm,

The fix is part of release **3.5.2**, which has been published a couple of days ago. Thank you again for having reported this issue.

With regards,

closed 2023-12-20 by ALABSTM