STMicroelectronics/stm32-mw-usb-host #13: STM32F401CC in USB HID Host not support all keyboards

AAJAY5, 2023-06-17:
# Issue
- I have generated USB HID Host code form STM32F401CC.
- Code is working perfect for some keyboard/scanners.
- I have attached descriptors of working and not working keyboard. And also init log from driver.
- Not working keyboard detected as keyboard but do not get data.

# Setup
[image]

- Working keyboard
    [image]

```
Information for device USB Device (VID=0x0483 PID=0x0011):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x000A
Current configuration value: 0x00
Number of open pipes: 0

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x0483	idVendor
0x0011	idProduct
0x0001	bcdDevice
0x01	iManufacturer   "USB Adapter"
0x02	iProduct        "USB Device"
0x00	iSerialNumber
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0022	wTotalLength   (34 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x01	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0110	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0049	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "USB Adapter"
0x02   0x0409  "USB Device"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
USB Device (VID=0x0483 PID=0x0011) Port: 4

Running on: Windows 10 or greater (Build Version 22621)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02

```

```
Information for device 2D Barcode Scanner (VID=0x1EAB PID=0x0003):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0007
Current configuration value: 0x00
Number of open pipes: 0

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x1EAB	idVendor
0x0003	idProduct
0x0000	bcdDevice
0x01	iManufacturer   "HENEX"
0x02	iProduct        "2D Barcode Scanner"
0x03	iSerialNumber   "00000000"
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xC0	bmAttributes   (Self-powered Device)
0xC8	bMaxPower      (400 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x01	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x0A	iInterface   "USB HID-Keyboard"

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0110	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x003F	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0x01	bInterval         (1 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x01	bEndpointAddress  (OUT endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor:
------------------------------
0x0C	bLength
0x03	bDescriptorType
Hex dump: 
0x0C 0x03 0x48 0x00 0x45 0x00 0x4E 0x00 0x45 0x00 
0x58 0x00 

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "HENEX"
0x02   0x0409  "2D Barcode Scanner"
0x03   0x0409  "00000000"
0x0A   0x0409  "USB HID-Keyboard"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
2D Barcode Scanner (VID=0x1EAB PID=0x0003) Port: 2

Running on: Windows 10 or greater (Build Version 19045)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02

```

- Not Working keyboard
    
    [image]

```
Information for device HIDKeyBoard (VID=0x26F1 PID=0x8801):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0008
Current configuration value: 0x01
Number of open pipes: 1

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x26F1	idVendor
0x8801	idProduct
0x0100	bcdDevice
0x01	iManufacturer   "TMS"
0x02	iProduct        "HIDKeyBoard"
0x03	iSerialNumber   "1234567890abcd"
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0022	wTotalLength   (34 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xE0	bmAttributes   (Self-powered Device, Remote-Wakeup)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x01	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0110	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x003F	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "TMS"
0x02   0x0409  "HIDKeyBoard"
0x03   0x0409  "1234567890abcd"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
HIDKeyBoard (VID=0x26F1 PID=0x8801) Port: 4

Running on: Windows 10 or greater (Build Version 22621)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02

```

ALABSTM, 2023-06-27:
ST Internal Reference: 156033

ALABSTM, 2023-06-27:
Hi @AAJAY5,

Thank you for this report. Your question has been forwarded to our development teams. We will get back to you as soon as we get their answer.

In the meanwhile, please note that STM32CubeF4 firmware still supports version **3.4.0** of this library while the latest version is **3.5.1**. Did you give it a try? You can easily integrate it into your package manually.

With regards,

ALABSTM, 2023-10-24:
Hi @AAJAY5,

How are you doing? Could you overcome the issue you described above? In the meantime, here is the feedback from our development teams:
* From the trace shared, the failing keyboard is well **recognized** and **enumerated** by our host. This means that the HID class driver has been **successfully started**.
* From that point on, without a trace from a protocol analyzer, it is hard to identify why the keyboard is not working as expected.
* Here is a **hint**:
  * Could you set a **breakpoint** at line `470` of the below part of the code to check that this line is reached after **pressing a key**?
  * If yes, this means that the function `USBH_LL_GetURBState()` returned `USBH_URB_DONE` upon the IN interrupt transfer request after a key has been pressed.

This info would help us better analyze this issue. Please let me know. Thanks.

https://github.com/STMicroelectronics/stm32_mw_usb_host/blob/f4332bd63dc43c251a95ae6a1eb1b15a6fcadaea/Class/HID/Src/usbh_hid.c#L467-L476

With regards,

ALABSTM, 2023-11-08:
Hi @AAJAY5,

I hope you are fine. Any reply please? Please allow me to close this issue as no activity has been registered since two weeks. Do not hesitate to reply if you would like the issue to be **reopened**.

With regards,

closed 2023-11-08 by ALABSTM