STMicroelectronics/stm32-mw-usb-host #15: Add the release version (Release vX.X.X) in the file header so the user can identify if his version is up to date

rtek1000, 2023-06-29:
There is no version information in the source code, in addition you could update the year too

```
/**
  ******************************************************************************
  * @file    usbh_cdc.c
  * @author  MCD Application Team
  * @brief   This file is the CDC Layer Handlers for USB Host CDC class.
  *
  ******************************************************************************
  * @attention
  *
  * Copyright (c) 2015 STMicroelectronics.
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
  *
  ******************************************************************************
  *  @verbatim
  *
  *          ===================================================================
  *                                CDC Class Driver Description
  *          ===================================================================
  *           This driver manages the "Universal Serial Bus Class Definitions for Communications Devices
  *           Revision 1.2 November 16, 2007" and the sub-protocol specification of "Universal Serial Bus
  *           Communications Class Subclass Specification for PSTN Devices Revision 1.2 February 9, 2007"
  *           This driver implements the following aspects of the specification:
  *             - Device descriptor management
  *             - Configuration descriptor management
  *             - Enumeration as CDC device with 2 data endpoints (IN and OUT) and 1 command endpoint (IN)
  *             - Requests management (as described in section 6.2 in specification)
  *             - Abstract Control Model compliant
  *             - Union Functional collection (using 1 IN endpoint for control)
  *             - Data interface class
  *
  *  @endverbatim
  *
  ******************************************************************************
  */
```

```
/**
  ******************************************************************************
  * @file    usbh_cdc.c
  * @author  MCD Application Team
  * @brief   This file is the CDC Layer Handlers for USB Host CDC class.
  *
  ******************************************************************************
  * @attention
  *
  * Copyright (c) 2023 STMicroelectronics (Release v0.0.0).
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
  *
  ******************************************************************************
  *  @verbatim
  *
  *          ===================================================================
  *                                CDC Class Driver Description
  *          ===================================================================
  *           This driver manages the "Universal Serial Bus Class Definitions for Communications Devices
  *           Revision 1.2 November 16, 2007" and the sub-protocol specification of "Universal Serial Bus
  *           Communications Class Subclass Specification for PSTN Devices Revision 1.2 February 9, 2007"
  *           This driver implements the following aspects of the specification:
  *             - Device descriptor management
  *             - Configuration descriptor management
  *             - Enumeration as CDC device with 2 data endpoints (IN and OUT) and 1 command endpoint (IN)
  *             - Requests management (as described in section 6.2 in specification)
  *             - Abstract Control Model compliant
  *             - Union Functional collection (using 1 IN endpoint for control)
  *             - Data interface class
  *
  *  @endverbatim
  *
  ******************************************************************************
  */
```

ALABSTM, 2023-06-30:
Hi @rtek1000,

Thank you for the proposal. However, such practice would make it very hard to identify updated files when visualizing the _diff_ between two consecutive versions.

Indeed, even though a file did not change from one version to the next, it would appear as such since the line containing the mention `(Release vX.Y.Z)` would have been necessarily updated.

I hope this clarifies why we prefer to avoid such practice. Please allow me to close this issue. Thank you for your comprehension and thank you again for your proposal.

With regards,

closed 2023-06-30 by ALABSTM

rtek1000, 2023-06-30:
Hi, yes, I agree that it is laborious. However, STM32CubeIDE uses tags to demarcate the code, I believe it wouldn't be too complicated to make a script to handle this. Thank you anyway.

rtek1000, 2023-06-30:
If there is interest in the future, I left a question in the GitHub community so that maybe they can give you some tips on how to read the Commit Message, which is already used to insert the version by your team:

https://github.com/orgs/community/discussions/59769

Thank you.